Bulebots/bulebule #84: Refactoring code structure

UserXGnu, 2017-10-03:
Well, the main.c file has a lot of function declarations as well as its implementations. To turn the code into a more clean and polished one, I believe that main.c should only have main() function and the respective includes as well as the main() function should not have to do much but call some handler to do what needs to be done. For that reason, I believe that main.c should be cleaned up and split into other files for better maintainablity. 

 I would like to aks forgiveness if I'm being mistaken or stupid about it...

=-D

Peque, 2017-10-04:
You are totally right. We will be splitting the code currently in `main.c` into many files as we continue development. We already started: see `common.c`, `detection.c`, `battery.c`...

Future changes in the code will include further code splitting. We agree with you that having a huge `main.c` is really ugly... :sweat_smile:

Peque, 2017-10-22:
I guess after 42abf9349456dbcd1e8dde2418e733ba8ab74217 we can close this issue. :wink:

closed 2017-10-22 by Peque