Bulebots/bulebule #177: Separate source and dependency files.

cua-cua, 2017-11-30:
Take out dependencies and objects of compilation from src folder to a new one. 
Create a new folder obj and modify makefile.

Peque, 2017-11-30:
What structure do you propose?

cua-cua, 2017-11-30:
At least ->
include: with headers (.h) 
obj:  with files from compilation ( .o and .d )

Peque, 2017-11-30:
I think it is more common to name the `obj` directory as `build`?

cua-cua, 2017-11-30:
With your proposal of build folder I realized ... what about .map and .elf files? 
So, ok for me with `build` for .o and .d files and .map, .elf and makefile ones outside these folders? (To compile and flash from the same directory?)

Peque, 2017-11-30:
I think they should all go in `build`, modifying the Makefile accordingly (don't know how :joy:).

Maybe we could create a top-level Makefile? To handle all tasks. In example:

- `make`: by default compiles the C sources.
- `make flash`: flashes the robot.
- `make sim`: compiles and runs client simulator.
- `make docs`: builds documentation and maybe even builds a local Python virtual environment with all dependencies if not already built. `pipenv`? :smile:
- `make ...`: other things maybe?