Bulebots/bulebule #255: Integrate gyroscope

Peque, 2018-04-14:
Configuration, calibration, integration in control (to measure angular velocity).

cua-cua, 2018-05-01:
- Create setup gyro function
- Create log function to connect gyro with spinete

Peque, 2018-05-01:
When I started playing around with the gyro I wrote this (do not know why or if it may be useful for your... :sweat_smile):

```c
	/* Reset gyro restoring default settings */
	gyro_write_register(GYRO_PWR_MGMT_1, 0x80);
	sleep_ticks(100);
	/* Reset signal path */
	gyro_write_register(GYRO_SIGNAL_PATH_RESET, 0x07);
	sleep_ticks(100);
	/* Set SPI mode, reset DMP + FIFO + SIG */
	gyro_write_register(GYRO_USER_CTRL, 0x1d);
	sleep_ticks(1000);
	printf("---\n");
	while (1) {
		reading = gyro_who_am_i();
		printf("I am %03i\n", reading);
		sleep_ticks(1000);
	}
```

Peque, 2018-05-01:
I think you might not need to configure anything in the gyro to read at 20 MHz.

You do need to make sure that, after gyro configuration, we change SPI frequency to ~20 MHz (as close as possible). We should be able to read at that speed.

Peque, 2018-05-01:
This might be useful as a reference:

- https://github.com/NordicPlayground/nrf52-quadcopter

Also, configuring a 6500 should be pretty similar (if not exact) to a 9250, so these might be useful as well:

- https://github.com/bolderflight/MPU9250
- https://github.com/brianc118/MPU9250

cua-cua, 2018-05-02:
Requirements:
- Maximum speed: 90º on 80 ms -> 1125º/s
- Minimum sampling: 1ms (1MHz)
- Read gyro on axe Z positive

closed 2018-05-14 by Peque