Bulebots/bulebule #418: Can't compile project

thekingisback-code, 2019-09-09:
When compiling the code (make) the following error is generated

`% make`
`sudo docker run -v $(pwd):/bulebule:Z -u $(id -u):$(id -g) bulebule make -s -C src/`
`make: find: Command not found`
`/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: invalid origin for memory` `region eeprom`
`collect2: error: ld returned 1 exit status`
`make: *** [opencm3/libopencm3.rules.mk:204: main.elf] Error 1`
`make: *** [src/main.elf] Error 2`

but make is correctly set up.
More info

`make --version`
`GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.`

`docker --version`
`Docker version 19.03.2, build 6a30dfc`

`rm-none-eabi-gcc --version`
`arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.`

Peque, 2019-09-15:
It says `make: find: Command not found`, so you may need to install `find` in your system. Or add it to the Dockerfile.

It also says `invalid origin for memory`. [Have a look at this issue](https://github.com/Bulebots/bulebule/issues/416), which probably fixes that.

Also, make sure you follow the ["Compiling" instructions in the documentation](https://bulebule.readthedocs.io/en/latest/brain.html#compiling). That means: `make docker` and `make libopencm3` before `make`. If you still get errors, report them back here. :blush:

thekingisback-code, 2019-09-15:
Hey there @Peque, I apologize but I'm new at this and its kind of hard debugging when I'm not really sure what I'm doing (and trust me, I'm googling everything I do).

> It says make: find: Command not found, so you may need to install find in your system. Or add it to the Dockerfile.

Find is already installed, I don't really know how to check for the version, but when I type `find --version` it returns 
```find: illegal option -- v
usage: find [-H | -L | -P] [-EXdsx] [-f path] path ... [expression]
       find [-H | -L | -P] [-EXdsx] -f path [path ...] [expression]
```
and while it doesn't output the version (wrong syntax I believe) which kind of means it's correctly installed.

Then I check where it is with `which find` and it outputs '/usr/bin/find'
With `echo $PATH` I verify the directory is inside.
`/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:/Library/Apple/bin:/opt/X11/bin`
I tried adding it to the Dockerfile just like `find \` (in the appropriate line, of course) and after running `make docker`this was the output

```
Complete!
Last metadata expiration check: 0:02:41 ago on Sun Sep 15 19:25:19 2019.
Package curl-7.65.3-3.fc30.x86_64 is already installed.
No match for argument: find
Error: Unable to find a match: find
The command '/bin/sh -c dnf update -y     && dnf install -y 	arm-none-eabi-gcc 	arm-none-eabi-newlib         curl         git 	make 	openocd 	python         wget         which 	find     && dnf clean all' returned a non-zero code: 1
make: *** [docker] Error 1

```
 
Still, I proceeded to compile libopencm3 using `make lipopencm3` and mid-compilation it outputted `Makefile:66: Your toolchain doesn't support -mcpu=cortex-m7.
Please use gcc-arm-embedded 4.8 2014q3 or newer. Skipping this sub-library.`which I suspected was because it was using the standard gcc on my system, which is 4.2.1. I added an Alias in the Dockerfile as follows `RUN alias gcc='g++-9'`which seemed logic at the moment, but it did not fix the problem.
Now I'm truly lost.

> It also says invalid origin for memory. Have a look at this issue, which probably fixes that.

```
I modified `src/stm32f103x8.ld` as the issue suggested and while it improved, now it outputs
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: main.elf section `.text' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: region `rom' overflowed by 5280 bytes
collect2: error: ld returned 1 exit status
make: *** [opencm3/libopencm3.rules.mk:204: main.elf] Error 1
make: *** [src/main.elf] Error 2
```

which to be honest, it could be carrying the errors from not completely compiling libopencm3, but I don't truly know.
Should I upload the files or something in specific?

Once again, I truly appreciate you taking the time to answer back.

Peque, 2019-09-15:
Does `make docker` output any errors? That is the first command that should succeed. I mean, once the image is correctly generated, everything else should work just fine.

Make sure to `git checkout -- Dockerfile` first, just to make sure you have a clean Dockerfile before starting. In fact, discard any changes you have locally to see if we can make at least the image generation work. :blush:

thekingisback-code, 2019-09-15:
Docker doesn't output any errors whatsoever.

```
sudo docker build -t bulebule .
Password:
Sending build context to Docker daemon  20.17MB
Step 1/6 : FROM fedora:30
 ---> e9ed59d2baf7
Step 2/6 : ENV LANG C.UTF-8
 ---> Using cache
 ---> c5fac37231e5
Step 3/6 : RUN dnf update -y     && dnf install -y 	arm-none-eabi-gcc 	arm-none-eabi-newlib         curl         git 	make 	openocd 	python         wget         which     && dnf clean all
 ---> Using cache
 ---> 7d562e0ad410
Step 4/6 : RUN alias python='python3'
 ---> Using cache
 ---> ec5ee188740c
Step 5/6 : VOLUME /bulebule
 ---> Using cache
 ---> 6f7250d9319e
Step 6/6 : WORKDIR /bulebule
 ---> Using cache
 ---> 20a47562eac9
Successfully built 20a47562eac9
Successfully tagged bulebule:latest
```
That's without running `git checkout -- Dockerfile`
So as far as I know, the image generation works @Peque

Peque, 2019-09-15:
Let us concentrate on the `make libopencm3` error then. Until that is fixed you should not try to compile the project.

Please, *do* execute `git checkout -- Dockerfile` to regenerate the clean/unmodified image from the Dockerfile included in Bulebule's repository.

Can you output the full output of the `make libopencm3` command once you regenerate the (clean/unmodified) image? Share it [through Gist](https://gist.github.com/), please.

thekingisback-code, 2019-09-15:
Okay, here's my complete terminal session @Peque 

```
WillyGuerrero@My-Macbook ~ % mkdir /Users/WillyGuerrero/Documents/CleanTry
WillyGuerrero@My-Macbook ~ % cd /Users/WillyGuerrero/Documents/CleanTry 
WillyGuerrero@My-Macbook CleanTry % git clone --recurse-submodules -j8 http://github.com/Bulebots/bulebule.git
Cloning into 'bulebule'...
warning: redirecting to https://github.com/Bulebots/bulebule.git/
remote: Enumerating objects: 28, done.
remote: Counting objects: 100% (28/28), done.
remote: Compressing objects: 100% (23/23), done.
remote: Total 2719 (delta 8), reused 14 (delta 5), pack-reused 2691
Receiving objects: 100% (2719/2719), 2.86 MiB | 478.00 KiB/s, done.
Resolving deltas: 100% (1934/1934), done.
Submodule 'src/mmlib' (https://github.com/Bulebots/mmlib.git) registered for path 'src/mmlib'
Submodule 'libopencm3' (https://github.com/libopencm3/libopencm3) registered for path 'src/opencm3/libopencm3'
Submodule 'printf' (https://github.com/mpaland/printf) registered for path 'src/printf'
Cloning into '/Users/WillyGuerrero/Documents/CleanTry/bulebule/src/mmlib'...
remote: Enumerating objects: 46, done.        
remote: Counting objects: 100% (46/46), done.        
remote: Compressing objects: 100% (30/30), done.        
remote: Total 217 (delta 24), reused 34 (delta 16), pack-reused 171        
Receiving objects: 100% (217/217), 66.98 KiB | 187.00 KiB/s, done.
Resolving deltas: 100% (115/115), done.
Cloning into '/Users/WillyGuerrero/Documents/CleanTry/bulebule/src/opencm3/libopencm3'...
remote: Enumerating objects: 39, done.        
remote: Counting objects: 100% (39/39), done.        
remote: Compressing objects: 100% (27/27), done.        
remote: Total 27415 (delta 15), reused 26 (delta 12), pack-reused 27376        
Receiving objects: 100% (27415/27415), 5.80 MiB | 414.00 KiB/s, done.
Resolving deltas: 100% (18225/18225), done.
Cloning into '/Users/WillyGuerrero/Documents/CleanTry/bulebule/src/printf'...
remote: Enumerating objects: 523, done.        
remote: Total 523 (delta 0), reused 0 (delta 0), pack-reused 523        
Receiving objects: 100% (523/523), 387.97 KiB | 272.00 KiB/s, done.
Resolving deltas: 100% (312/312), done.
Submodule path 'src/mmlib': checked out '48fc8df47a83c4e15ae742d73127d5d78e1c5b7d'
Submodule path 'src/opencm3/libopencm3': checked out '38125e9941ad9dba0fdbcc762dc4bdcd59b7c8c4'
Submodule path 'src/printf': checked out '0dd4b64bc778bf55229428cefccba4c0a81f384b'
WillyGuerrero@My-Macbook CleanTry % cd bulebule
WillyGuerrero@My-Macbook bulebule % make docker
sudo docker build -t bulebule .
Password:
Sending build context to Docker daemon  20.17MB
Step 1/6 : FROM fedora:30
 ---> e9ed59d2baf7
Step 2/6 : ENV LANG C.UTF-8
 ---> Using cache
 ---> c5fac37231e5
Step 3/6 : RUN dnf update -y     && dnf install -y 	arm-none-eabi-gcc 	arm-none-eabi-newlib         curl         git 	make 	openocd 	python         wget         which     && dnf clean all
 ---> Using cache
 ---> 7d562e0ad410
Step 4/6 : RUN alias python='python3'
 ---> Using cache
 ---> ec5ee188740c
Step 5/6 : VOLUME /bulebule
 ---> Using cache
 ---> 6f7250d9319e
Step 6/6 : WORKDIR /bulebule
 ---> Using cache
 ---> 20a47562eac9
Successfully built 20a47562eac9
Successfully tagged bulebule:latest
WillyGuerrero@My-Macbook bulebule % git checkout -- Dockerfile
WillyGuerrero@My-Macbook bulebule % make libopencm3
sudo docker run -v $(pwd):/bulebule:Z -u $(id -u):$(id -g) bulebule scripts/setup_libopencm3.sh
--2019-09-15 22:00:19--  https://raw.githubusercontent.com/libopencm3/libopencm3-examples/master/examples/rules.mk
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.180.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.180.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
[... 573 lines cut ...]
  CC      vector.c
  CC      systick.c
  CC      scb.c
  CC      nvic.c
  CC      assert.c
  CC      sync.c
  CC      dwt.c
  AR      libopencm3_samd.a
  BUILD   lib/vf6xx
  CP      libopencm3_vf6xx.ld
  CC      ccm.c
  CC      uart.c
  CC      gpio.c
  CC      iomuxc.c
  CC      vector.c
  CC      systick.c
  CC      scb.c
  CC      nvic.c
  CC      assert.c
  CC      sync.c
  CC      dwt.c
  AR      libopencm3_vf6xx.a
make: Leaving directory '/bulebule/src/opencm3/libopencm3'
WillyGuerrero@My-Macbook bulebule % 
```

thekingisback-code, 2019-09-15:
Heres the complete session of my terminal, @Peque .

https://gist.github.com/thekingisback-code/fcce31e9c81b97d7f57998b1e1ec2b22

Peque, 2019-09-15:
Ok, it does not look too bad! ^^

I do not understand why you get that error when compiling for the F7 family (I do not, and theoretically, using the Docker image, we should be compiling `libopencm3` with the same toolchain). :confused: 

Anyway the F7 family is not required to build the project so:

- Can you share the output of `make`
- Also the output of `docker --version`

thekingisback-code, 2019-09-15:
> Ok, it does not look too bad! ^^
There's hope! lol

Anyway, here's the new output, @Peque (Should I keep on tagging you?, btw)
https://gist.github.com/thekingisback-code/f56e97c9d96af12dd3e9771b85c37e23

Peque, 2019-09-15:
Okay, I do not know much about Docker, but I find it weird that you get different outputs than me. I, for instance, can compile for the F7 family and I do not get the "find: Command not found" error... :confused:

You now need to apply the fix in https://github.com/Bulebots/bulebule/issues/416. That is for sure.

Try replacing the `arm-none-eabi-gcc` with `arm-none-eabi-gcc-cs-7.4.0` in the Dockerfile and see if that helps (share the output of `make`).

You can also try adding `findutils` to the list of packages installed in Dockerfile and try to `make` again? (share the output again).

Peque, 2019-09-15:
No need to tag me now. Once I start posting in an issue, I get automatically subscribed. :wink:

thekingisback-code, 2019-09-15:
In lines `875-876` it now outputs the error after applying the fix in [#416](https://github.com/Bulebots/bulebule/issues/416) and in line `869` a new error is introduced, which to be honest looks promising, sounds easier to fix.
But who am I to say, at this point, I don't really understand that much.

My full terminal session is [here](https://gist.github.com/thekingisback-code/3098c79f7da8813c9de2de23b36792b4) and the new Dockerfile is [here](https://gist.github.com/thekingisback-code/7d9b3fc80c76018846aa8a8881881bb9)

Peque, 2019-09-15:
Yeah, [that line](https://gist.github.com/thekingisback-code/3098c79f7da8813c9de2de23b36792b4#file-thingy2-txt-L869) looks like maybe you need to clean the (pre)compiled libopencm3 files. Maybe you should try to reset the project (i.e.: create the directory, clone, make libopencm3 and make again). Not very efficient, but maybe the safest way to ensure we are on the right track. :joy:

Peque, 2019-09-15:
Or maybe just `rm -rf ./src/opencm3/` and then `make libopencm3`. Becayse if you reset the project, you will need to edit the Dockerfile and the `.ld` file again...

thekingisback-code, 2019-09-15:
Took the long way home and reset the project (created the directory, cloned, make docker, make libopencm3 and make again...) and this is the output, now it only overflows, pretty sure it's related to [fix #416](https://github.com/Bulebots/bulebule/issues/416).

[Here's](https://gist.github.com/thekingisback-code/51303dc7d8f6562bd20b43624eb86c39) the terminal session.

thekingisback-code, 2019-09-15:
Somehow I closed it (whatever it means), but... nope.

Peque, 2019-09-15:
I can now reproduce the error, which is good news! :blush:

I think I was using my (old) cached Docker image, which probably had different packages installed.

This means we may be able to fix this by setting the version of other package(s) in the Dockerfile. I would say the package we need to set to an older version is probably `arm-none-eabi-newlib`. But not sure which version it should be. :sweat_smile:

thekingisback-code, 2019-09-15:
Anything I can do to help on my end?
Seriously dude, I owe you a coffee/beer.

Peque, 2019-09-15:
Yeah, mostly because I am about to disconnect, and I do not think I will be able to spend much time with this until, probably, next Sunday. :innocent: 

Have a look at the [newlib Koji packages](https://koji.fedoraproject.org/koji/packageinfo?packageID=17579). Look for the `.fc30` packages (Fedora 30).

I would first try with [the 3.0.0 version](https://kojipkgs.fedoraproject.org//packages/arm-none-eabi-newlib/3.0.0/5.fc30/noarch/arm-none-eabi-newlib-3.0.0-5.fc30.noarch.rpm). Install it with:

```
dnf install arm-none-eabi-newlib-3.0.0-5.fc30.noarch.rpm
```

You will need to edit the Dockerfile again to include this package installation in your image. I would add a new `RUN` line before the `RUN alias`, so you do not need to regenerate the image from zero.

thekingisback-code, 2019-09-16:
Worry not, I'll try to do it all by myself and I'll update the issue as necessary mid-week.

> You will need to edit the Dockerfile again to include this package installation in your image. I would add a new RUN line before the RUN alias, so you do not need to regenerate the image from zero.

This doesn't sound that hard, so I'll try!

Thanks!

Peque, 2019-09-16:
Actually, it seems that may not be necessary. Simply set `arm-none-eabi-newlib-3.1.0-2` in the Dockerfile and try. :blush:

Peque, 2019-09-16:
* I meant `arm-none-eabi-newlib-3.1.0-2.fc30`. Otherwise `dnf` will not find the package.

Peque, 2019-09-16:
@thekingisback-code Please, try these changes and tell me if that works fine: https://github.com/Bulebots/bulebule/pull/420

closed 2019-09-16 by Peque