Bulebots/bulebule #419: Constants and whatnot

thekingisback-code, 2019-09-09:
In _config.h_

How are the locomotion related constants calculated?
(My biggest issue/question is with WHEELS_SEPARATION and SHIFT_AFTER_180_DEG_RUN? 
I believe I already know how to get MICROMETERS_PER_COUNT. 
How are the value of the constants SENSOR_SIDE_... determined?
What's the main difference between the constants MOUSE_WHEELS_SEPARATION and WHEELS_SEPARATION?

In _setup.h_

How are you measuring your mouse? (Mouse dimensions)
In case modifications are donde to the main PCB, how are the exact dimensions measured?

Peque, 2019-09-15:
We need to update the documentation. :sweat_smile: 

- `WHEELS_SEPARATION`: Not really used now. It was used before, when we were not using the gyroscope, to estimate the angular velocity based on the encoders readings. It is an experimental value (i.e.: it is set based on experience/trials).
- `SHIFT_AFTER_180_DEG_RUN`: When your mouse turns in-place, the position of the "center" of the mouse usually is translated a bit. In our case, the translation only occurred in one direction (frontal, not lateral displacement), so this represents the frontal displacement. It is an experimental value as well.
- `SENSOR_[SIDE|FRONT]_[LEFT|RIGHT]_[A|B]`: we calibrate the front sensors with the `run front_sensors_calibration` command (sent over bluetooth using the `script/connect_bluetooth.py`). We save the logs and calculate the constants with `scripts/notebooks/front_sensors_calibration.ipynb`). The side sensors are not really calibrated, we just use the same, or similar constants to those gotten from the front sensors.
- `MOUSE_WHEELS_SEPARATION` is used to estimate the forces on the mouse while running and estimating the feed-forward constants.

Mouse dimensions (all viewed from the top):

- `MOUSE_MOTOR_TAIL`: measured from the motor's axis to the back-end of the PCB.
- `MOUSE_MOTOR_HEAD`: measured from the motor's axis to the front-end of the PCB.
- `MOUSE_AXIS_SEPARATION`: measured from the front wheels' axis to the back wheels' axis.
- The other constants are calculated base on those 3.

thekingisback-code, 2019-09-15:
> We need to update the documentation. 😅

If all goes well, I could help with that. ☺️

Anyway, isn't `WHEELS_SEPARATION` used in `update_encoder_readings()` and in `inplace_turn()`? And well, those are still used. Am I a little bit confused?
Or are you talking about an unreleased version?

Peque, 2019-09-15:
Good catch! :blush:

Yeah, it is used in `update_encoder_readings()` to update the `angular_speed` variable in the `encoder` module. That variable, though, would only be used if there was a call to `get_encoder_angular_speed()`, but there is none (there seems to be one in the `logging` module, but that is not really used for running).

As of today, the angular speed is read from the gyroscope. Only if you did not want to use the gyro (or if you did not have one) you would need to call `get_encoder_angular_speed()`.

thekingisback-code, 2019-09-15:
Oh la la, got it.
Thanks! 
After a week of reading your team code (and don't get me wrong, its amazing) I'm finally getting the hang of it! 😅

Peque, 2019-10-30:
Removed old encoder-based angular speed code to avoid confusion: https://github.com/Bulebots/bulebule/commit/f58a0e7abcaaceb7649490b3ab81eda7409286b9