Bulebots/bulebule #428: Replace Blue Pill?

Peque, 2019-10-30:
https://www.aliexpress.com/item/4000103610226.html

rafacouto, 2019-11-10:
We, teraghases team, have worked in a F4 draft, if you were interested in: https://github.com/RecunchoMaker/teraghases/blob/master/doc/adefesio32/stm32f446.pdf :wink:

Peque, 2019-11-10:
@rafacouto Thanks for sharing! Looks good! :blush:

We would like to keep basing our Bulebule design on easily-available and already-soldered mini-PCBs, so if we replace the current Blue Pill it will probably be with another prototyping board. Note that in Bulebule the motor driver, the gyroscope, the DC-DC converters, the microcontroller... they are all on their respective prototyping PCBs. That makes the robot easier to solder and reduces the bill of materials.

We also have a new design (Meiga) based on the F4! :blush: You can find the schematics/PCB here:

https://upverter.com/design/bulebots/meiga/

rafacouto, 2019-11-11:
Meiga seems to be very competitive and its name is magical 😉  Bookmarked.
Happy micromousing! 👍