Bulebots/bulebule #431: Software (pre compiled binaries)

ishwariyjoshi, 2019-12-26:
Can i flash precompiled binaries of this code if yes so, can another give me that? Pre compile .bin file.

Peque, 2019-12-26:
What issues have you found trying to compile the code yourself?

I am assuming you already have a Bulebule built and ready. We would love to know about it! :heart_eyes: Do you have pictures to share? Have you made any modifications? What are you going to use the robot for? :blush:

ishwariyjoshi, 2019-12-27:
@Peque  sure, I like to share my work. but it's still under progress we are building a custom Bot. basically I just replaced some components and as per my 💵  budget. I thought i can replaced the motor with "N-20 motor + encoder 1000RPM." and other circuits as same as 'bulebule' design. but I'm don't have any idea from where o start the software setup that's why I'm asking for the pre-compiled binaries.😉

Peque, 2019-12-29:
Sounds good! I would love to see some pictures of your robot. Can you share them here? :blush:

If you are having trouble compiling the project, could you share the specific issues you are finding? Have you read [the documentation](https://bulebule.readthedocs.io)?

ishwariyjoshi, 2020-01-11:
hi, @Peque I ❤️ Bulebots Design. with the help of this as my reference design, I built my own bot and I also win the competition and secured my second position, thanks to your design which helps me lots. I always😍 this kind of project as an open-source project. your work is amazing guys keep it up. and I also like to share some photographs of my bot.  
[image]
[image]
[image]

Peque, 2020-01-11:
@ishwariyjoshi Thanks for your kind words and thanks for sharing some pictures! It looks great!! :blush: 

As a suggestion for the next upgrade, you may want to add some ecoders (see the [GA12-N-20 with integrated encoders](http://es.aliexpress.com/item/33016910240.html)). They are not super expensive and they help a lot with speed control. :wink:

I will close this issue, since I understand the documentation is clear enough on how to compile and flash the code.

Good luck with your micromouse project and keep us updated with your advances! We love to know about successful micromouse stories based on Bulebule. :heart_eyes:

Peque, 2020-01-11:
@ishwariyjoshi Also, the battery looks big and heavy. Consider using a 2-cell battery instead, with smaller capacity (240 mAh is probably enough, maybe 500 if you really want to play with your robot a lot before charging or if you do not have a battery replacement). Reducing weight is always a good thing.

closed 2020-01-11 by Peque