Bulebots/bulebule #432: Image building fails

thewires2, 2020-03-18:
Makefile:5: *** recipe commences before first target.  Stop.

I keep getting this error . Can someone please suggest what is to be done?

Peque, 2020-03-18:
@thewires2 Did you follow the [instructions in the documentation](https://bulebule.readthedocs.io/en/latest/software.html#compiling)?

Are you using Docker/Podman as suggested?

thewires2, 2020-03-19:
> @thewires2 Did you follow the [instructions in the documentation](https://bulebule.readthedocs.io/en/latest/software.html#compiling)?
> 
> Are you using Docker/Podman as suggested?

I did follow the according to the instructions given
[image]

thewires2, 2020-03-19:
> @thewires2 Did you follow the [instructions in the documentation](https://bulebule.readthedocs.io/en/latest/software.html#compiling)?
> 
> Are you using Docker/Podman as suggested?

I have followed the steps exactly as mentioned . I don't understand why is this error taking place
[image]

thewires2, 2020-03-19:
> @thewires2 Did you follow the [instructions in the documentation](https://bulebule.readthedocs.io/en/latest/software.html#compiling)?
> 
> Are you using Docker/Podman as suggested?

I am using podman and i have followed the steps as mentioned. Please see if i am making any error in the terminal
[image]

Peque, 2020-03-19:
It seems the `Makefile` is failing with:

```
make: command: Command not found
```

If you look at the `Makefile` you can see, in line 4:

```makefile
ifeq (, $(shell command -v ${ENGINE}))
```

Basically, it is calling `command` to see if `podman` is available in the system. It seems you do not have the `command` command... :joy: 

Thought `command` was available in all POSIX-like shell environments. Which OS/distro are you using? Out of curiosity.

I just pushed some changes that will hopefully fix this issue for you. Make sure to pull from latest `master`.

thewires2, 2020-03-19:
I am using Linux with Ubuntu 19.10. I pulled the code from the latest master and got a new error. Is it an issue with my system?? 
[image]

thewires2, 2020-03-19:
> I am using Linux with Ubuntu 19.10. I pulled the code from the latest master and got a new error. Is it an issue with my system??
> [image]

I searched stack overflow for this particular error but was unable to find it 🙁

Peque, 2020-03-19:
@thewires2 What is the output of `podman info`?

Peque, 2020-03-19:
You may need to populate the configuration files in Ubuntu:

https://computingforgeeks.com/how-to-install-podman-on-ubuntu/

thewires2, 2020-03-19:
> You may need to populate the configuration files in Ubuntu:
> 
> https://computingforgeeks.com/how-to-install-podman-on-ubuntu/
> 
> In particular, you will need to add at least one registry to search for images in your `/etc/containers/registries.conf`. You can add a couple of popular ones:
> 
> ```toml
> [registries.search]
> registries = ['docker.io', 'quay.io']
> ```

It worked!! Thank you so much 😃 😃

closed 2020-03-19 by thewires2