Bulebots/bulebule #437: 3D mount file has error

Cognisant1, 2021-05-15:
UsingCQ-editor, the Mount does not work properly.  Before the circle points are added, the workplane needs to state (centerOption="CenterOfMass") otherwise the motor Mount points are on the edge of the board rather than the centre.

Peque, 2021-05-15:
It may be due to changes in the latest 2.1. Before, I had seen:

```
cadquery/cadquery/utils.py:21: FutureWarning: Default walue of centerOption will change in the next relase to ProjectedOrigin
```

Don't know which was the previous default `centerOption`.

Peque, 2021-05-15:
Yeah, it was changed in 2.1:

https://github.com/CadQuery/cadquery/blob/master/changes.md

@Cognisant1 Want to create a PR with the changes? :wink:

Cognisant1, 2021-05-15:
Hi. I’ve got the version of the file that works but this is my first action in GitHub so I don’t know how to post it. Any help would be great!

Peque, 2021-05-15:
@Cognisant1 Maybe this is a good starting point:

https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/proposing-changes-to-your-work-with-pull-requests

:blush:

Cognisant1, 2021-05-16:
Here is the modified cadquery mount.py. I couldn't get GitHub to copy my file so I've given the code itself - sorry if incorrect!
`"""
This script defines the mount for the wheels and motor.
"""
from math import asin
from math import cos
from math import sin
from math import radians

import cadquery

def circle_points(number, circle, shift=0):
    radius = circle / 2.
    step = 360 / number
    points = [(radius * sin(radians(i * step + shift)),
               radius * cos(radians(i * step + shift)))
              for i in range(number)]
    return points

# Gears
MODULE = 0.3
Z_PINION = 15
Z_GEAR = 60

# Motor
MOTOR_SHIFT = 1.5
MOTOR_DIAMETER = 15
MOTOR_HOLE_DIAMETER = 6.5
MOTOR_MOUNT_THICK = 1.5
MOUNT_MINIHOLES_CIRCLE = 10
MOUNT_MINIHOLES_DIAMETER = 1.7

# Mount
MOUNT_THICK = 6
MOUNT_HEIGHT = MOTOR_DIAMETER
MOUNT_WIDTH = 40
MOUNT_FILLET = 1

# Holes
SCREW_SPACE = 34
SCREW_DIAMETER = 2
AXIS_DIAMETER = 3

# Basic mount structure
mount = cadquery.Workplane('XY').box(MOUNT_WIDTH, MOUNT_HEIGHT, MOUNT_THICK)

# Base screws
mount = mount.faces('<Y').workplane()\
    .pushPoints([(-SCREW_SPACE / 2., 0), (SCREW_SPACE / 2., 0)])\
    .hole(SCREW_DIAMETER)

# Motor holes
miniholes = circle_points(number=6, circle=MOUNT_MINIHOLES_CIRCLE)
mount = mount.faces('<Z').workplane(centerOption="CenterOfMass")\
    .pushPoints(miniholes)\
    .hole(diameter=MOUNT_MINIHOLES_DIAMETER)
mount = mount.faces('>Z').workplane().cboreHole(
    MOTOR_HOLE_DIAMETER,
    MOTOR_DIAMETER + 1,
    MOUNT_THICK - MOTOR_MOUNT_THICK)

# Axis
axis_shift = (MODULE * (Z_PINION + Z_GEAR)) / 2.
axis_shift *= cos(asin(MOTOR_SHIFT / axis_shift))
mount = mount.faces('<Z').workplane()\
    .pushPoints([(axis_shift, -MOTOR_SHIFT), (-axis_shift, -MOTOR_SHIFT)])\
    .hole(AXIS_DIAMETER)

# Fillet
mount = mount.edges('|Z').fillet(MOUNT_FILLET)
`

Cognisant1, 2021-05-16:
That was very messy - sorry
I've now added the modified mount.py file (and the mount.stl and rim.stl files, in case they'd be useful) in to this zip folder.i hope that's better practice!
 [Mount_OK.zip](https://github.com/Bulebots/bulebule/files/6488872/Mount_OK.zip)

Peque, 2021-05-16:
@Cognisant1 I'd say, in general, it is better to share your code changes by either:

- Sharing the code diff in a comment, if it is just a couple of lines (you can execute `git diff` and copy-paste the output)
- Pushing your changes to a branch in your fork and sharing that branch or, even better, opening a pull-request with the changes! :wink: 

Do you want to try with the pull-request?

Cognisant1, 2021-05-16:
Yes that would be good, but I’m still not sure how to open a pull request. Do you think the stl files are worth adding??

Peque, 2021-05-16:
@Cognisant1 I'd rather not add the STL files. Instead, I would prefer to pin the CadQuery version in the environment, to make sure 3D designs are reproducible and do not break with future CadQuery updates.

Cognisant1, 2021-05-18:
Hi Miquel

Is there a reason why you didn't 3D print the gear onto the rim. It would avoid centring errors.

Peque, 2021-05-18:
@Cognisant1 Indeed!

The reason was that we got some [good second hand motors](https://bulebule.readthedocs.io/en/latest/building.html#motors-and-encoders) really cheap from Aliexpress (at the time they were 8 or 15 € each!), so we designed the locomotion system around them. The motors came with a 0.3 modulo gear soldered to the shaft and we didn't have access to a precise-enough 3D printer, so we couldn't print wheel gears that were compatible with the motor...

Nowadays, with resin 3D printing you could print even a 0.3 modulo gear I think, so it should be easier! :blush: 

Please, share any modifications you make to the locomotion system! We'd love to know about different designs that do not require the currently hard-to-find-second-hand (or otherwise too expensive) Faulhaber motors. :wink:

Peque, 2021-05-18:
Yeah, the Faulhaber encoders are great. :stuck_out_tongue_winking_eye:  

There are a couple of references in Alibaba (https://spanish.alibaba.com//trade/search?fsb=y&IndexArea=product_en&CatId=&SearchText=1524B009SR), but I think I once contacted them and they did not have any available in reality... Maybe you have a higher chance with [the more expensive seller](https://spanish.alibaba.com/product-detail/the-motor-new-original-1524b009sr-60821956259.html?spm=a2700.galleryofferlist.normal_offer.d_title.2eb61b99lePJF3), if it fits within your budget.

Anyways, software can do magic so, even if it will be harder, you can achieve a good performance with much worse encoders. Also, the Futura mouse is a great reference, but you may find this project interesting (with pretty [easy-to-find encoders](https://github.com/ukmars/ukmarsbot/wiki/Fitting-Encoders)):

https://github.com/ukmars/ukmarsbot/wiki

:wink:

Cognisant1, 2021-05-18:
Yes. I've already built a ukmarsbot. It's good but not very fast🙁.  Peter Harrison is a great software and electronic genius. He managed to double the pololu encoder count using a quad XOR gate and his profilers give a very smooth ride. The Nano is slow and a bit limited in memory. I liked your blue pill and the gyro. Maybe I just need to save up for the Faulhaber motors 😣

Peque, 2021-05-19:
@Cognisant1 Ah, not your first micromouse then! :smile: 

Yeah, the Bluepill is enough for not-very-complex control algorithms. I also started some work with a new Bluepill-like F4 board (https://github.com/Peque/bulebule/commit/2b3799c71168a3d3fbc35a4173a083a511ced357), but I would not consider it as stable... Only recommended if you want to have some fun with hardware design and testing! :stuck_out_tongue_winking_eye: 

Here is a fork of the Bulebule project with a different locomotion system:

https://github.com/hmightypirate/AllMyCircuits/wiki/Buriburi-(Micromouse)

I think the motors are a little too weak, but maybe you can take it as an inspiration for your own design. A very common design is with the motors between the wheels and the encoder at the front/back wheel's shaft. Note that those motors are quadcopter's motors, very cheap and with high power. Encoders are cheap too, but require a little bit of work to design and mount.

Here I did some research for the new board (and with new encoders):

https://github.com/Bulebots/bulebule/issues/428

The as5047p looked like a great contender. :wink:

Cognisant1, 2021-05-19:
Wow, Miguel

What a feast of wonderful information!! Thank you so much😁

I'll have fun digesting all this. 😊

Cognisant1, 2021-06-11:
Hi Miguel

Really sorry to bother you again, but it seems impossible to source the 5v and 9v boost boards to fit the PCB layout. All the boards now seem to have the ground as the middle pin whereas yours has the ground at one end. Can you please indicate where these boards came from or send me the Eagle (or equivalent files) so that I can slightly modify the PCB layout? Modifying the gerber files themselves is really difficult!

You’ve been so kind in your help so far that I hope I can ask for this extra assistance!

Best wishes

Adrian

Peque, 2021-06-11:
@Cognisant1 No problem! :blush: 

- [9 V converter](https://es.aliexpress.com/item/32647774026.html) (pack of 10... we donated the spare parts to a local "fablab" to avoid waste, or you can try to find single pieces :wink:)
- [5 V converter](https://es.aliexpress.com/item/32752700862.html)
- [3.3 V converter](https://es.aliexpress.com/item/32774098119.html) (again, pack of 5...)
- [Motor driver](https://es.aliexpress.com/item/32709948530.html)
- [Gyro board](https://es.aliexpress.com/item/1937917136.html)
- [Bulebule schematics and PCB design](https://upverter.com/design/bulebots/bulebule/), feel free to fork and make any modifications you want! :blush: 

Feel free to add this info to the project documentation. I didn't want to sponsor Aliexpress but... it may be useful for people trying to build the robot to at least have one option to buy the parts.

Cognisant1, 2021-06-11:
That’s brilliant, Miguel. Thank you so much. 

Adrian

Cognisant1, 2021-07-02:
Hi Miguel

I’ve decided to try to get the faulhaber motors even though they are so expensive. However I have had this reply from one of the suppliers:
“ Unfortunately, Faulhaber do not offer the motor encoder with the pinion you require”

Can you help at all? 

Best wishes

Adrian

Sent from my iPhone

Peque, 2021-07-02:
@Cognisant1 If you are thinking about buying brand new motors, maybe you should consider the 1717 instead (I think they are about the same price). Those are the ones we wanted to use for [our Meiga design](https://upverter.com/design/bulebots/meiga/). They have a couple of advantages:

- They allow for a slimmer robot design 
- They have a higher no-load speed (14 000)
- Maybe others (don't remember :joy:)

Think whether you want the 3 V version or the 6 V version... I think @micromouseonline used the 3 V version. You'll need less battery cells in series to drive the motor but you'll have higher current peaks. If you have a look at [2018 All Japan contestants' hardware](http://www.ntf.or.jp/mouse/micromouse2018/recode2018_181215.pdf) you'll find many Faulhaber with both 3 V and 6 V versions so... Both are fine I guess. Note the extended use of the 1717 and no 1524.

You can buy the pinion/gears from another shop (this one is [an example here in Spain](http://www.mootio-components.com/_refe_008472.html)). Here again you need to decide you preferred modulo. Bulebule uses 0.3, but @micromouseonline uses 0.5 or 0.6 (if I remember correctly) and got 2nd place in 2017 All Japan! So not that you need lower modulo to run faster. :stuck_out_tongue_winking_eye: 

Also, if you are doing a "tetra" design, think about the whole design of the locomotion system (tires, gears, pinion, PCB distance to floor...). It all goes together. We documented [a quick guide about tetra locomotion systems](https://bulebots.readthedocs.io/en/latest/tetra.html) if you want to have a look. You can easily [visualize your own configuration with Python](https://github.com/Bulebots/bulebots/blob/master/docs/source/tetra.py).

Last but not least, make sure you are asking your supplier for the motor plus the encoder! (a Faulhaber motor by default comes with no encoder). If you forgot that... the new price will probably double. :sweat_smile: 

If you would rather make your own encoders, then I think Faulhaber motors are not worth it. Just use common quadcopter motors instead, which are much cheaper. :wink:

micromouseonline, 2021-07-02:
If your goal is top performance use the 1717 motors with encoders but 
you may be shocked at the price. I use 3V parts and drive them from a 2S 
LiPo battery with a maximum drive voltage of 6.6Volts.

My gears are 0.5mod because they are cheap, easy to find and more 
tolerant of positional errors.